oscam-emu/oscam-emu #282: Please modify the simulation

atefganm, 2018-08-29:
2018/08/29 22:32:45 13B649BD r   (reader) emulator [emu] dvbapiau emmtype=unique, len=155 (hex: 0x9B), cnt=1: written (24 ms)
2018/08/29 22:32:45 13B649BD r   (reader) emulator [emu] dvbapiau emmtype=unique, len=101 (hex: 0x65), cnt=1: error (0 ms)
2018/08/29 22:32:52 13B649BD r      (emu) A new hash mode [16] is in use.
2018/08/29 22:32:52 697AC5EB c        (-) -- Skipped 7 duplicated log lines --
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2018/08/29 22:32:52 697AC5EB c        (-) -- Skipped 8 duplicated log lines --
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR): Invalid argument
2018/08/29 22:32:52 697AC5EB c      (ecm) dvbapiau (0E00@000000/0000/0008/A2:27B59F524602E69BD3CF4FD7D07B69A9): found (3 ms) by emulator - PIX India
2018/08/29 22:32:52 13B649BD r      (emu) A new hash mode [16] is in use.
2018/08/29 22:32:52 697AC5EB c        (-) -- Skipped 7 duplicated log lines --
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2018/08/29 22:32:52 697AC5EB c        (-) -- Skipped 8 duplicated log lines --
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2018/08/29 22:32:52 697AC5EB c   (dvbapi) ERROR: ioctl(CA_SET_DESCR): Invalid argument
2018/08/29 22:32:52 697AC5EB c      (ecm) dvbapiau (0E00@000000/0000/0008/A2:85E34924A28DD289C2EC8FED82207FDD): found (3 ms) by emulator - PIX India

caveman99, 2018-08-29:
There will be no change public until the inner workings of Hashmode 16 are known.

atefganm, 2018-08-29:
Thank you Thomas Göttgens

We are waiting for the amendment

scolanzi, 2018-09-04:
Caveman99 is this also true for hashmode 17?

caveman99, 2018-09-04:
Yes, currently only Hash Mode 1-10 are know. Anything larger than 10 is unknown.

FCS001FCS, 2018-09-09:
Newbie here.

Is it that hard to determine the Hash Modes from a stream? Can someone explain the process? Maybe we can assist?

I have no idea how it is done but am really missing AFN on 9E.

caveman99, 2018-09-10:
IF you look at the source code, the higher hash modes are essentially an XOR table. The computed control data is combined (XOR) with fixed values. The hard part is to detemine the table for a new hash mode, sonce there's more than one XOR per CW Value. Some of the lower hash modes actually use computational algorithms. This can not easily be determined fom comparing actual output to exprected output, cause you don't know the expedted output that will actually descramble the signal.

FCS001FCS, 2018-09-10:
Thanks for the explanation. Way beyond my knowledge but I hope it is being worked on by someone with the skills needed.

jlmariense, 2018-09-27:
here in South America ja three satellites have entered hash mode 17

scolanzi, 2018-09-28:
Now, Mux ESPN BR  40,5W (SES6) in hash mode 17 too.

oscam-emu, 2018-10-15:
Partially fixed with r772 https://github.com/oscam-emu/oscam-emu/commit/9e07fbc848895c9c62af24137c9c0771294aec52

closed 2018-10-15 by oscam-emu