oscam-emu/oscam-emu #284: audio not working on some channels

PeterMaxu, 2018-10-03:
audio not working on some channels without streamrelay on vti image

oscam-emu, 2018-10-03:
Give us more info about your problem and a livelog.

PeterMaxu, 2018-10-03:
audio remain silent, with exact the same config and streamrelay on, audio is ok.
for example: 43°W 3865 H Hola TV no audio, golf channel audio ok
what debug switches you want me to set for the log?

my oscam.conf:

[global]
logfile                       = /tmp/oscam.log
nice                          = -1
preferlocalcards              = 2
block_same_ip                 = 0
block_same_name               = 0

[cache]
max_time                      = 8

[streamrelay]
stream_relay_enabled          = 0

[dvbapi]
enabled                       = 1
au                            = 1
pmt_mode                      = 0
delayer                       = 60
ecminfo_type                  = 1
user                          = dvbapiau
read_sdt                      = 1
write_sdt_prov                = 1
extended_cw_api               = 1
boxtype                       = dreambox

nautilus7, 2018-10-03:
I don't receive these channels, but I think it's a driver/hardware issue, not emu's fault. What is your box?

A livelog won't hurt (debug on), but I don't think I can do much anyway...

oscam-emu, 2018-10-03:
I don't receive these channels, but I think it's a driver/hardware issue, not emu's fault. What is your box?

A livelog won't hurt (debug on), but I don't think I can do much anyway...

PeterMaxu, 2018-10-05:
> A livelog would be welcome (debug ALL), but I don't think I can do much

Its a VU+ Solo4k

live log will follow...

lorba7068, 2018-10-20:
This case very similar to the 121w and several channels example Dynamite TyCs A3Series attached logs.
[121w-A3SeriesNTCS.txt](https://github.com/oscam-emu/oscam-emu/files/2498544/121w-A3SeriesNTCS.txt)
[121w-CinemaDinamita.txt](https://github.com/oscam-emu/oscam-emu/files/2498545/121w-CinemaDinamita.txt)
[121w-ViendoMoviesHd.txt](https://github.com/oscam-emu/oscam-emu/files/2498546/121w-ViendoMoviesHd.txt)

oscam-emu, 2018-10-20:
I'm afraid this is a issue with Vu+ receivers. Do you have another receiver brand to test?

lorba7068, 2018-10-20:
Probe some time ago with a dreambox 800 SE and another friend of mine with a dm920 and the same problem with some channels, some time ago "joeuser" helped with a Fix for AC3 and corrected a large part but there were some channels that continue with problems such as those I describe.

jlmariense, 2018-10-21:
Friend, I was following the conversation. Below is my livelog from oscar:
---> 2018/10/21 18:55:26 0D7A26E5 c      (ecm) dvbapiau (0E00@000000/0000/0070/7E:A5E4D4D3FA62EDCBF45BAA7E96218301): found (8 ms) by emulator - Disney HD Bra
2018/10/21 18:55:27 5F5FCCE1 r      (emu) Key found in EMM: P 0039**** 00 1E45D4D086B423 ; UA: 0059EDF7
2018/10/21 18:55:27 5F5FCCE1 r      (emu) Key found in EMM: P 0039**** 01 929FFDC790612F ; UA: 0059EDF7
2018/10/21 18:55:27 5F5FCCE1 r   (reader) emulator [emu] dvbapiau emmtype=unique, len=155 (hex: 0x9B), cnt=1: written (44 ms)
2018/10/21 18:55:27 5F5FCCE1 r   (reader) emulator [emu] dvbapiau emmtype=unique, len=155 (hex: 0x9B), cnt=1: error (1 ms)

After the new oscam, only original audio

joeusercz, 2018-10-22:
Need full logs starting from BEFORE channel is tuned. 
Stream relay uses software descrambling instead of the built in hardware deccramblers. Drivers for HW descramblers have limits as to how many pids can be descrambled at one time. Actually, there are two limits, one for the number of pids per channel and a second limit for the total number of pids. Obviously different drivers for different boxes have different limits. The option for "Max pids for extended CWs" is used to limit the number of pids per channel to allow possibly more channels to be descrambled at the same time without reaching the limit of the total number of pids. 
But, we need the full logs including the PMT parsing to see if this is the problem.

lorba7068, 2018-10-22:
Hello Joeusercz, here are these logs I hope to help correct this problem. Thank you in advance for taking time and helping us.
[ViendoMov_PMT.TXT](https://github.com/oscam-emu/oscam-emu/files/2503477/ViendoMov_PMT.TXT)
[HolaTv_PMT.TXT](https://github.com/oscam-emu/oscam-emu/files/2503478/HolaTv_PMT.TXT)
[A3HD_PMT.TXT](https://github.com/oscam-emu/oscam-emu/files/2503479/A3HD_PMT.TXT)
[A3Hd-121w.txt](https://github.com/oscam-emu/oscam-emu/files/2503480/A3Hd-121w.txt)
[HolaTv_121w.txt](https://github.com/oscam-emu/oscam-emu/files/2503481/HolaTv_121w.txt)
[ViendoMov_121w.txt](https://github.com/oscam-emu/oscam-emu/files/2503482/ViendoMov_121w.txt)

joeusercz, 2018-11-01:
Those are not full logs - need to see how oscam parses the pmt, not dvbsnoop...

oscam-emu, 2018-11-02:
@joeusercz how can I contact you in private? Can you email me at the mail written on the readme file? Thanks!