oscam-emu/oscam-emu #288: Not yet open channels with hash mode 17

jlmariense, 2018-10-19:
After leaving oscam with hash mode 17 and 18, it generates normal channel keys that was in hash 17. but does not open image. What can it be?

Log:
2018/10/19 12:16:19 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:DD152365BBDCCDF26C04E76292CDC89F): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:20 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:4126AFAEAD07AD311A65C276B6C28294): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:21 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:8C296855651CF46861570EAF523672E2): found (8 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:22 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:A670F32336236125AF9831533DF8DE65): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:23 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:3EB312463682ADDD791D45C79764EAEB): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:24 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:2C180B42961721B3CA0DA6C162A64707): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:25 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:1E58FF3AC0414665CE21E512E5BD41C0): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:26 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:CF95B5AFF6E82FBA07385CA6EE07781E): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:27 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:0F85B3373E8B2D68D09F211E99D0405B): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:28 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:A0B3634892DE2D93BFFBAA482C774642): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:29 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:CD78C266637F5FD9EF8A8E2729148A4B): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:30 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:2FAC27B1B629C3A836D1E7453D7BF91A): found (7 ms) by emulator - Sportv 1 HD
2018/10/19 12:16:31 5319E705 c      (ecm) dvbapiau (0E00@000000/0000/00A1/48:9318AEF182AC034E96FE13422FBE549B): found (7 ms) by emulator - Sportv 1 HD

oscam-emu, 2018-10-19:
It seems to be a problem and channels using hash mode 18 don't clear. We are investigating...

engineerkhan15, 2018-10-20:
Yes, I can confirm that Sony Network at Asiasat 7 (105.5 East) is not clearing with this fix, although it is using hash mode 18.

oscam-emu, 2018-10-20:
Test with r774 https://github.com/oscam-emu/oscam-emu/commit/d8f4f3d7ff4c4cbe67da014a05ee95322689aba2 if it is fixed

jlmariense, 2018-10-20:
Wonderful. I am available to test if you need

jlmariense, 2018-10-21:
This is Livelog:
2018/10/21 02:02:56 2DFED9AD c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 0E00 PROVID 000000 ECMPID 1771 ANY CHID PMTPID 1429 VPID 0262
2018/10/21 02:02:57 2DFED9AD c      (ecm) linuxsatsupport (0E00@000000/0000/00A1/48:D2378A407F2B0A0C03E24F8CEBFF747B): rejected group (1 ms) - Sportv 1 HD (no matching reader)
2018/10/21 02:02:58 2DFED9AD c      (ecm) linuxsatsupport (0E00@000000/0000/00A1/48:4F46E9BE0412CF7FFCC84A7178222610): rejected group (0 ms) - Sportv 1 HD (no matching reader)
2018/10/21 02:02:58 2DFED9AD c   (dvbapi) Demuxer 0 restarting decodingrequests after 0 ms with 1 enabled and 0 disabled ecmpids!

budiarno, 2018-10-21:
Thank you. Confirmed it fixes the SPN.

Wonderful.

soomarali, 2018-10-21:
linuxsatsupport (0E00@000000/0000/0005/76:2EC37F47E97F16BEFC6490E93E19CB6E): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:43 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:DED89665E20AD67827501FC5C97B751E): found (6 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:44 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:4FBFC2F8B860D159FB311D6734F5D5F3): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:45 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:B13B79AE683E8481F47F763EC33958D0): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:46 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:152522C519E4F03A519D4383EFDFA22C): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:46 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:331AF5C3A7E85A65604D7C2FE15977B2): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:47 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:D1388EF379CD69D67852B1AF8D2ED8BA): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:48 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:4CC4DE2AE40B9D4DC254BBA90189026C): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:49 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:F84618A883B40231F0B5EB32DDFF0E5B): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:50 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:E9B6D7A7A0451F681E30D83A5E3AD21C): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:50 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:8D406D94818C8B3119989195CB78C66C): found (6 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:51 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:34E129E620AB969BA4B2B5C34062F98A): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:52 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:A29793960B1040E58ED712C6E7ABEDB7): found (6 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:53 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:E741031F0BD7E988EE6F3641CE099870): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:54 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:F610BB51BF21F543466A1E8533C64F46): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:54 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:A9BAD10A3F57E104F083B6E16BF016C1): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:55 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:943B714F986047524C5D7487FCD08D2D): found (6 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:56 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:2324CD8A9FDA648BE9FD225BDAC9C046): found (5 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:57 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0005/76:C0E3C9C4184347884926A3F028E44954): found (8 ms) by emulator (L/1/4/4) - HBO India
2018/10/21 07:34:57 4FF25ED1 c   (dvbapi) Demuxer 0 stop descrambling program number 0005 (HBO India)
2018/10/21 07:34:59 4FF25ED1 c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 0E00 ECM_PID: 1774 PROVID: 000000 
2018/10/21 07:34:59 4FF25ED1 c   (dvbapi) Demuxer 0 ecmpid 1 CAID: 09DD ECM_PID: 1B9F PROVID: 000000 
2018/10/21 07:34:59 4FF25ED1 c   (dvbapi) Demuxer 0 found 2 ECMpids and 6 STREAMpids in caPMT
2018/10/21 07:34:59 4FF25ED1 c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 0E00 PROVID 000000 ECMPID 1774 ANY CHID PMTPID 13CF VPID 0BFF
2018/10/21 07:34:59 4FF25ED1 c   (dvbapi) Demuxer 0 found 2 ECMpids and 6 STREAMpids in PMT
2018/10/21 07:34:59 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0047/7E:7E67EEA8696CD94B721049ED070257AD): found (5 ms) by emulator (L/1/4/4) - LePLEX HD
2018/10/21 07:34:59 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0047/7E:FFD6354E50CA920BA6BA1ED0220857B1): found (67 ms) by emulator (L/1/4/4) - LePLEX HD
2018/10/21 07:35:03 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0047/7E:266B4EF8990E3C752E5359E4FD07FE5B): found (4 ms) by emulator (L/1/4/4) - LePLEX HD
2018/10/21 07:35:13 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0047/7E:A1BF0F2BD7D381803FD604E717E1AF94): found (3 ms) by emulator (L/1/4/4) - LePLEX HD
2018/10/21 07:35:23 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0047/7E:BAEF56A0E8FE4D17FD8F536E2E7B7B8A): found (4 ms) by emulator (L/1/4/4) - LePLEX HD
2018/10/21 07:35:33 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0047/7E:F35740676C47031F28DA10FC0E9B9B2F): found (3 ms) by emulator (L/1/4/4) - LePLEX HD
2018/10/21 07:35:43 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0047/7E:B363BD0BEFE09E5AB75752029CDD04D7): found (3 ms) by emulator (L/1/4/4) - LePLEX HD
2018/10/21 07:35:53 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0047/7E:AD3E1B503C015F6EE0C2759D44DEF38C): found (3 ms) by emulator (L/1/4/4) - LePLEX HD
2018/10/21 07:36:03 4FF25ED1 c      (ecm) linuxsatsupport (0E00@000000/0000/0047/7E:86E66ED6DDA4A9A433996D943AF665F8): found (3 ms) by emulator (L/1/4/4) - LePLEX HD

engineerkhan15, 2018-10-21:
Yes, I also confirm that it is working now. Thanks a lot developers, for your help.

jlmariense, 2018-10-21:
Good morning, I wanted to thank the friends involved in the project. Oscam11436 R774, perfect opening. Thank you so much for your work.