oscam-emu/oscam-emu #289: A new hash mode [20] is in use

engineerkhan15, 2018-10-24:
Hi Developers,
Today, Sony PowerVu channels at Intelsat 20 (68.5 East) started using hash mode 20. Can you add fix for this hash mode in your patch?

jlmariense, 2018-10-30:
No friend, here is the way hash mode 21
2018/10/30 15:43:57 17982683 r      (emu) A new hash mode [21] is in use.
2018/10/30 15:43:57 3B3779A6 c        (-) -- Skipped 7 duplicated log lines --
2018/10/30 15:43:57 3B3779A6 c      (ecm) dvbapiau (0E00@000000/0000/0089/48:081E64EB36BCD67E1CB6F5AC469F911F): found (3 ms) by emulator - Premiere FC HD

budiarno, 2018-11-02:
Solved with latest commit of hashmode 19 and 20 here:
https://github.com/oscam-emu/oscam-emu/commit/b6df0605d7ef0987e42b24cb4a92744363262d70

closed 2018-11-07 by oscam-emu