oscam-emu/oscam-emu #290: With emu r775 black screen on PowerVu SIS Channels

periccapio, 2018-11-06:
With emu r775 - Oscam r11438 black screen on PowerVu SIS Channels (Astra 4A 12418 H 27500).
Note that emu r774 - Oscam r11436 works OK

oscam-emu, 2018-11-06:
Did you change your reader to not ignore cw checksums for powervu?

See here: https://github.com/oscam-emu/oscam-patched/wiki/PowerVU-config#november-2018-update---attention

periccapio, 2018-11-06:
Nope,sorry.I apologize for it.But AFN channels work without disablecrccws_only_for = 0E00:000000 directive

oscam-emu, 2018-11-06:
No problem.

AFN use the CSA algorithm for stream encryption, so they actually use every forth byte as checksum.
Other PowerVu channels use DES, so we must tell OSCam that there are no checksum bytes in the CW.